wpilibsuite/wpilib-ws-robot-romi #31: Restarting the program from VSCode doesn't restart the program on the robot

bradamiller, 2020-11-21:
I have a program which only runs the motors in auto periodic.
1. Do "Simulate robot code on desktop" and start the robot going.
2. Select auto in the UI and the motors start going.
3. Do "Simulate robot code on desktop" again from VSCode without stopping the robot program

The program isn't stopped, the motors keep running, and a second UI instance is started.

jpokornyiii, 2020-11-30:
I'm not able to reproduce. I assume you are referring to the halsim_gui extension. When I follow the steps, the second build fails because of "...\HalTest\build\tmp\jniExtractDir\opencv_core347.dll (The process cannot access the file because it is being used by another process)"

Therefore no new program starts (killing the old one) and no new GUI. I'm fairly new to the halsim_gui though. Is there another step I need to configure first?

jpokornyiii, 2020-11-30:
Please note, upon close of the first window the program ends and disables the robot. My gut feeling here is that this issue is with the halsim_gui but will need to reproduce to confirm.

jpokornyiii, 2020-12-03:
@bradamiller , what platform are you running your halsim_gui on?

jpokornyiii, 2021-03-02:
Closing for now. If the issue is still persisting, please re-open.

closed 2021-03-02 by jpokornyiii