wpilibsuite/wpilib-ws-robot-romi #67: Yellow LED Default State

jasondaming, 2020-12-28:
The Yellow LED is active HIGH.  The default state is also HIGH but the LED is off.  In order to turn the LED on  you need to first set the LED to LOW then back to HIGH.

I would take the expected operation one step further however and since all of the other LEDS are active LOW then we should probably abstract this difference in the library.  Just so that when the user calls set<Color>LED they all operate the same way so the user doesn't have to know the active HIGH vs LOW difference.

zhiquanyeo, 2020-12-29:
All the LEDs should be active HIGH (i.e. `digitalOut.set(true)`). I'll take a look at this. There was a race condition in earlier versions of the Romi code that led to some DIO messages getting dropped prior to initialization. It's possible that the patch I made fixed this.

jasondaming, 2020-12-29:
I was going off this: https://www.pololu.com/docs/0J69/3.2

zhiquanyeo, 2020-12-29:
Ah ok. The underlying firmware uses the `ledRed`, `ledGreen` and `ledYellow` functions as described here: https://pololu.github.io/romi-32u4-arduino-library/_romi32_u4_8h.html

These already do the abstraction of HIGH = LED On

zhiquanyeo, 2021-01-07:
@jasondaming Can you try out the latest version of the Romi software (1.0.3) and see if this still happens? It's packaged with the latest version of the WPILibPi image

jasondaming, 2021-01-09:
It will be at least a couple days as I lent my Romi to a student....maybe I will get a different one tomorrow?  I'll respond back when I know

jasondaming, 2021-01-22:
I just ran this and I can't remember for sure but I think before the other LEDs were okay and it was just the Yellow that has this issue.  It looks like Yellow has been brought in line with the others but the wrong way.  Now all of the LEDs have this issue.  That is if I want to turn on an LED I have to call setGreenLED(false) and then setGreenLED(true) because the state is high to start (but the LED is not illuminated.

jpokornyiii, 2021-03-10:
Reproduced the issue. I will work on debugging it.

closed 2021-03-16 by zhiquanyeo