wpilibsuite/wpilib-ws-robot-romi #75: Motor and Encoder Direction Phasing

icemannie, 2021-01-01:
Note only. Current positive encoder direction makes the front of the robot the pi connector one.

Issue:-

The motor encoders count positive up when a negative level signal is being applied to the motors. Closing a velocity or position loop won’t work unless some negating is done.

The Romi Reference example ‘works’ because it doesn’t negate the gamepad Y value as is typically done.

Inverting both encoders makes it more logical but flips the robot “front”. That might easily be the case on a robot with encoders separate from the drive motors.

zhiquanyeo, 2021-01-01:
That's actually correct. The front of the robot is the one with the suspension arm caster (also where the Raspberry Pi USB ports are). By convention, negative drive values correspond to forward motion, since forward stick on a joystick results in a negative value on that axis.

icemannie, 2021-01-01:
It’s been common practice to negate the Y joystick value in the teleop
routine.

icemannie, 2021-01-01:
From the simulation code used in the Hour of Code video showing Y from controller being negated. Also the PID can just use the output values without signs.
 @Override
  public void autonomousPeriodic() {
    m_pidController.setP(SmartDashboard.getNumber("Kp", 10.0));
    m_pidController.setD(SmartDashboard.getNumber("Kd", 0.0));

    double output = m_pidController.calculate(m_drive.getDistanceMeters(), kSetpoint);
    m_drive.drivePercent(output, output);
  }

  @Override
  @SuppressWarnings("LocalVariableName")
  public void teleopPeriodic() {
    double xSpeed = -m_controller.getY(Hand.kLeft);
    double rot = -m_controller.getX(Hand.kLeft) * 0.4;

    m_drive.drivePercent(xSpeed - rot, xSpeed + rot);
  }

icemannie, 2021-01-01:
The Romi Demo done on the Hour of Code video has inverted encoders 
leftEncoder = new Encoder(Constants.leftEncoderA, Constants.leftEncoderB, true);
and with a + command in the Forward routine shows the robot moving with the pi connectors at the rear of the robot. The joystick Y teleop value isn't inverted so the robot moves in the opposite direction than expected and counts negative.

PR #76 (2021-01-02): Flip sign on motor input values so that +ve = forward
Resolves #75
closed 2021-01-02 by zhiquanyeo