wpilibsuite/wpilib-ws-robot-romi #87: Encoder Glitches

roboticsgeek, 2021-01-18:
Team 1073 Here. Love the MiniBots.

We were having a lot of trouble getting closed loop controllers to behave with the robot (PID velocity controller), so we started digging into the data and it seems like every so often the encoder feedback is "glitching" for lack of a better term.  Attached is a converted shuffleboard log showing the encoder values we get back. We modified the example program to send a constant 40% PWM to each motor and the Romi is "up on blocks" with its wheels free in the air touching nothing.

Note in the logs places like Rows: 312-314 where Cols: AE/AG show an encoder flipping sign and briefly moving *backwards* though the motor is rolling smoothly forwards. This happens fairly often on both the left and right sides of our Romi. Once every few seconds.

This glitch makes any kind of closed loop control very difficult since the motors seem to "jump" randomly at the control loop will respond to / amplify this jump.

[recording-12.59.37.xlsx](https://github.com/wpilibsuite/wpilib-ws-robot-romi/files/5831890/recording-12.59.37.xlsx)

Rows 375/376 Cols Z/AA show the other encoder (that is moving in the other direction) has the same kind of "go backwards" issue. This was typical of what we were fighting with because it is reporting reversing to 2x the original speed in a single sample... which is physically impossible.

The logging format only catches a fraction of the # of times this happens to our encoders which you can see by watching the data fields in the simulator window and/or shuffleboard, etc. you see the velocity randomly flip signs by a large amount in a single sample and then flip back.

Any help appreciated on getting reliable encoder data from our MiniBot!

roboticsgeek, 2021-01-18:
This may also be a bug or issue in Romi firmware or WPILibPi-Rombi build...

zhiquanyeo, 2021-01-18:
Interesting. I just did a similar test to what you have and did notice a little bit of glitching, although not as bad as in your data. It looks like we're getting some of the glitchy data during the data read from the Romi, which would narrow it down to somewhere in firmware, or something going wrong in the I2C connection. I'll take a closer look to see what else I can find.

zhiquanyeo, 2021-01-18:
Did a little more digging and on the firmware side, it looks like all is well (I had some debug statements printed out over serial that would flag if the encoder values did not move in the expected direction e.g. should increase if the motor input is forward). Looks like this is glitching at the I2C connection. I'll see if I can figure something out.

Having the robot on the ground (or at least with some resistance against the wheels) does reduce the glitching a little (with the same "drive forever at 40%" program)

roboticsgeek, 2021-01-23:
Just updating our status here. We updated our Romi to the new WPILibPI "Romi" release 2021.2.1 and the problem persists (perhaps even worse). We're basically blocked on using the robot at this point. The first thing we're trying to do is closed loop velocity control of the drivetrain and the glitches are so bad and so persistent they just blow apart any code that tries to use feedback control at this point. 

Perhaps it is some additional information that we get the glitches in "blocks" where it glitches several times in a row and then have a period of 0.5 - 1 second where things work fine and then "glitch block". It's like everything is fine until the motors get up to about 1/2 speed and then glitch glitch glitch constantly.

Can we try to get logging off something else that will help diagnose the problem.

zhiquanyeo, 2021-01-23:
Do you have a link to some of the code that you were using to test? We'll try to get some extra logging on both sides of the software/firmware stack shortly.

roboticsgeek, 2021-01-23:
I'll try to make a specific example test on a branch for you. Additional information that might be helpful. Today we got a second Romi robot kit from Pololu and set it up just like the first one. It is running an RPI 4 which is the only difference. The issue persists and is exactly the same with the 2nd set of hardware, so I'm pretty sure that is some kind of software issue somewhere in the stack. I'll try to make a specific / simplified test program for it.

icemannie, 2021-01-25:
[image]

This is a Shuffleboard capture of one side Romi encoder running at constant speed. There is no scaling in the encoder. The readings are being taken in the periodic method of the drive subsystem.
The left chart is the encoder rate reading the right is the position.
The speed shows significant spikes but around 4000 counts per second seems to be the flat sections.

roboticsgeek, 2021-01-25:
Thanks for the data analysis and additional information.

I was looking at the Romi WS robot and how it interfaces with the Romi over I2C. I've not looked into the code deeply, but just looking at the data types and considering the "shared memory" nature of the problem I was wondering if this could be some kind of race condition?

If I'm updating an int16_t over I2C, but doing it 1 byte at a time... then I can introduce glitches if I don't complete the read in one block and then update the shared memory fully all at once. Just a thought to consider for folks who know more about it.

zhiquanyeo, 2021-01-26:
Could either of you download this: https://github.com/wpilibsuite/wpilib-ws-robot-romi/suites/1898552610/artifacts/37203132

Unzip it (you should end up with a .tgz file) and then upload the .tgz file using the Web UI->Romi->Update Web Service section.

This fixes an (admittedly boneheaded) bug with the i2c readWord method and should give much better results. I tried it with a simple "run at 40%" test, and didn't see the glitches I was seeing earlier

icemannie, 2021-01-26:
[image]
Left and right speeds and position. Big spikes are clearly no longer there.
Looks like this  can go on the Fixed side.

closed 2021-01-26 by jpokornyiii