wpilibsuite/wpilib-ws-robot-romi #94: Romi stops taking input (possibly related to interrupted wifi)

michaelhill, 2021-02-10:
This is a result of this ChiefDelphi thread: https://www.chiefdelphi.com/t/romi-randomly-stops-taking-input/391320

Basically, I went through the getting started guide, flashed my Raspberry Pi 3B with the WPILibPi 2021.2.1 image, and have Web Service 1.1.4 running. I created a Java example RomiReference, and run the debug (Run -> Start Debugging). The simulation GUI started just fine. I'm able to switch to Teleop mode and drive the robot around with my joystick with no problem. However, it seems when Wifi connection gets spotty (either I'm driving it a bit if a distance ~40 ft, or a wall is between my laptop and the Romi), sometimes it seems like the Romi will just hold the last input values. This means if I was driving forward when it "disconnects", it continues going forward. Sometimes it will recover after several seconds, and play through the "buffer" of inputs with some delay. But a large portion of the time, even if I go chase down the Romi, and physically bring it next to my computer (where there should be no connection problems), the wheels will still be spinning until I physically turn it off.

Not sure if this is related, but some warnings also pop up around the time of these issues:

`Warning at edu.wpi.first.wpilibj.IterativeRobotBase.printLoopOverrunMessage(IterativeRobotBase.java:301): Loop time of 0.02s overrun

Warning at edu.wpi.first.wpilibj.Tracer.lambda$printEpochs$0(Tracer.java:63):   teleopPeriodic(): 0.030736s
        SmartDashboard.updateValues(): 0.000054s
        robotPeriodic(): 0.000134s
        LiveWindow.updateValues(): 0.003910s
        Shuffleboard.update(): 0.000004s
        simulationPeriodic(): 0.000004s`

Additionally, I've also seen this error come up once in a while, which could also be related:

`HALSimWS: Websocket Disconnected
TCP connection closed`

More testing will need to be done to positively determine which commands directly relate to losing connection.

However, all this said, my expectation is that all motors should cut off if there is a watchdog timeout or connection is lost.

jpokornyiii, 2021-02-10:
Can you try the latest release here? 
https://github.com/wpilibsuite/wpilib-ws-robot-romi/releases/tag/v1.1.4

You can update the firmware via the web interface http://wpilibpi.local

Please let us know if this resolves your issue.

jpokornyiii, 2021-02-11:
We are discussing one element that's not protected yet and that is delayed messaging between CPU and PI. The latest version should protect against runaway motors on disconnect and disablement.

jpokornyiii, 2021-03-02:
Closing this issue due to implementation in the linked PR and https://github.com/wpilibsuite/wpilib-ws-robot/pull/26. Leveraging the constant messaging from the driver station we are able to determine if we are still receiving messages from wpilib. If we don't receive a message in 500 ms we will kill the heartbeat to the Romi which is checked every 1 sec setting a max overrun of 1.5 sec.

closed 2021-03-02 by jpokornyiii