wpilibsuite/wpilib-ws-robot-romi #97: Seeing a lot of drift in the romi

willsnow, 2021-02-18:
A little history: 
I started with the romi-example code, made sure all firmware was up to date, etc. Romi service version is 1.1.4, shows as "firmware compatible" (is there an easy way to figure out the firmware version?). Calibrated the gyro several times and we experience some rather signficant drift. Putting in 90 for a turn can result in widely varying actual turns - up to 30deg. I've never once seen it turn what looked like 90. Turning right, it's 100-130. Turning left it's 50-70.

Next, I read zhiquanyeo's article on frc-characterization. I followed the instructions and after some experimentation, I got that working and generated the analysis files. I'm still showing a track width of .6887466822 which seems ridiculous - the track width of the robot is _not_ 27+ inches. Nevertheless, I plugged in the values into the example ramsete trajectory code, and to make things a little simpler, I changed the path to just go straight .75M, turn left and go .75M then stop.

List.of(
          new Pose2d(0d, 0d, new Rotation2d(0)),
          new Pose2d(0.75, 0.75, Rotation2d.fromDegrees(90))

Results: 

16" straight ahead, turns and goes 30". The great news is the angle of the turn was mostly correct, and it's repeatable. 
Changed track width to .1524 (6" - measured track of romi), now it goes 30" out, turns left 45 deg and maybe 11". Again repeatable. 

Is there some trick to picking track width? Or do I just binary search for a value that mostly works and live with it?

I also notice in the dashboard with the romi sitting still, the angle_x, angle_y and angle_z on the romigyro are just ever increasing which seems weird.

zhiquanyeo, 2021-02-19:
The firmware doesn't actually have a "version" per se. Instead it has an identifier that's re-generated whenever the interface between the Raspberry Pi and 32U4 board changes (when major changes happen on the firmware, we also re-generate the identifier to force an update). 

Re: trackwidth - I'll try running that example on a couple of robots I have on hand. Earlier versions of the frc-characterization tool had a bug where the trackwidth value wasn't getting calculated correctly, so make sure you've updated that tool as well.

What rate is the angle_z increasing at? The hardware itself does have a bit of drift, and we do a couple of software tricks to smooth it out (some done as part of calibration), but you will definitely see some drift in the long term.

willsnow, 2021-02-19:
angle_z is ever increasing (although in one case it was ever decreasing, making it extra special). maybe 1 degree per 20 seconds.

I'm happy to try things out. I might have repeatability now (going to check again today, see if everything is the same as yesterday or I have to do a full new characterization). While not perfect, consistency is better than nothing.

One additional note: do you find a lot of difference when changing batteries? As in I swapped from LiON batteries to standard alkalines and the robot characterization was different. 

I'm happy to share any data you might need. The version I have of frc-characterization is now 2021.2.2; I'll rerun the characterization and see if the numbers are different (which is likely - I've never gotten the same answer twice)

zhiquanyeo, 2021-02-19:
That drift isn't too bad actually (0.05dps - My well behaved bots have similar drift ranges (0.01 - 0.05dps)). The LSM6DS33 has noise in the +/-0.85 degree per second range (we do a little bit of onboard smoothing to attempt to get this down a little bit more).

What voltages were the LiON batteries at? (Nominal voltage that is) We normalize voltage calculations on the Romi to a 9V range (which is what 6x 1.5V alkalines give you), and that information gets scaled again to a 12V scale for voltage reporting to robot code. If you're driving the wheels via voltage (-12 to 12V) commands, the behavior should be the same across battery types. I'll double check the behavior in a bit.

willsnow, 2021-02-24:
I'm seeing about 8.6 volts on the LiON batteries, but regular alkaline batteries are way less - like high 6 volts

willsnow, 2021-02-24:
Also, as I mentioned over in your frc-characterization code project, I'm getting zero function on the characterization since I updated.

zhiquanyeo, 2021-03-29:
This should now be resolved with v1.3.0. If this is still an problem, please put in another issue. Thanks!

closed 2021-03-29 by zhiquanyeo