wpilibsuite/wpilib-ws-robot-romi #98: Visual Studio Code

4420Robot, 2021-02-24:
Our installation of Visual Studio Code  on startup says 
BUILD FAILED in 1s
<-------------> 0% WAITING
> IDLE
The terminal process "cmd.exe /d /c gradlew build generateVsCodeConfig   -Dorg.gradle.java.home="C:\Program Files\Java\jdk-15.0.2"" terminated with exit code: 1.
We have no other information and no idea why it's happening.

PeterJohnson, 2021-02-24:
Java 11 is required, the message points to Java 15.  Did you install using the WPILib installer?

4420Robot, 2021-02-26:
> Java 11 is required, the message points to Java 15. Did you install Visual Studio Code and all dependencies using the WPILib installer?

 Configure project :
INFO: Could not find files for the given pattern(s).

> Task :assemble FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':assemble'.
> No buildable binaries found:
    - executable 'frcUserProgram:debug:executable':
        - No tool chain is available to build for platform 'windowsx86-64':
            - Tool chain 'visualCpp' (Visual Studio):
                - Could not locate a Visual Studio installation, using the command line tool, Windows registry or system path.
            - Tool chain 'gcc' (GNU GCC):
                - Could not find C compiler 'gcc' in system path.
            - Tool chain 'clang' (Clang):
                - Could not find C compiler 'clang' in system path.
            - Tool chain 'roborioGcc' (RoboRioGcc):
                - Don't know how to build for platform 'windowsx86-64'.
    - executable 'frcUserProgram:release:executable':
        - No tool chain is available to build for platform 'windowsx86-64':
            - Tool chain 'visualCpp' (Visual Studio):
                - Could not locate a Visual Studio installation, using the command line tool, Windows registry or system path.
            - Tool chain 'gcc' (GNU GCC):
                - Could not find C compiler 'gcc' in system path.
            - Tool chain 'clang' (Clang):
                - Could not find C compiler 'clang' in system path.
            - Tool chain 'roborioGcc' (RoboRioGcc):
                - Don't know how to build for platform 'windowsx86-64'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 44s
1 actionable task: 1 executed
The terminal process "cmd.exe /d /c gradlew build generateVsCodeConfig   -Dorg.gradle.java.home="C:\Users\Public\wpilib\2021\jdk"" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

So we used the WPIlib installer and have the correct version of java now, we got a whole new error, any idea how to fix it?

jpokornyiii, 2021-03-02:
Looks like it can't find your C++ compiler. Do you know where it was installed and if it's on your PATH environment variable? If you run "Git Bash" you can leverage the command "which gcc". Try the installer again and ensure that "C++ Compiler" is checked. Let me know if this helps.

jpokornyiii, 2021-03-10:
Assuming the issue is with installation. Please try updating your PATH env var to include the location of your Compiler or try a new install.

closed 2021-03-10 by jpokornyiii