wpilibsuite/wpilib-ws-robot-romi #113: Romi Crashes when adding duplicate NT keys

Daltz333, 2021-04-30:
Accidentally was updating a Field2D in a loop

```java
  @Override
  public void autonomousPeriodic() {
    if (timer.get() < trajectory.getTotalTimeSeconds()) {
      var desiredPose = trajectory.sample(timer.get());

      prevPoses.add(desiredPose.poseMeters);

      Drivetrain.field.getObject("DesiredPose").setPoses(prevPoses);
```

where prevPoses is an array of poses.

This causes a crash found in the wpilibpi log

```
/home/pi/.nvm/versions/node/v14.15.0/lib/node_modules/@wpilib/wpilib-ws-robot-romi/node_modules/node-ntcore/dist/protocol/v3/v3-nt-client.js:453
            throw new Error(`Client has an existing key ${msg.entryName}`);
            ^

Error: Client has an existing key /SmartDashboard/Field/DesiredPose
    at V3NTClient._handleEntryAssignment (/home/pi/.nvm/versions/node/v14.15.0/lib/node_modules/@wpilib/wpilib-ws-robot-romi/node_modules/node-ntcore/dist/protocol/v3/v3-nt-client.js:453:19)
    at V3NTClient._handleData (/home/pi/.nvm/versions/node/v14.15.0/lib/node_modules/@wpilib/wpilib-ws-robot-romi/node_modules/node-ntcore/dist/protocol/v3/v3-nt-client.js:412:34)
    at RRSocket.<anonymous> (/home/pi/.nvm/versions/node/v14.15.0/lib/node_modules/@wpilib/wpilib-ws-robot-romi/node_modules/node-ntcore/dist/protocol/nt-client.js:48:18)
    at RRSocket.emit (events.js:315:20)
    at Socket.<anonymous> (/home/pi/.nvm/versions/node/v14.15.0/lib/node_modules/@wpilib/wpilib-ws-robot-romi/node_modules/node-ntcore/dist/transport/rr-socket.js:123:18)
    at Socket.emit (events.js:315:20)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:284:9)
    at Socket.Readable.push (_stream_readable.js:223:10)
    at TCP.onStreamRead (internal/stream_base_commons.js:188:23)
```

zhiquanyeo, 2021-04-30:
Looks like this is caused by a type switch from array to raw when we exceed 255/3 number of poses. This is supported by the NT3 protocol but wasn't implemented in the NT library used by the Romi. I'll fix this soon-ish

jpokornyiii, 2022-10-28:
I built a test harness and reproduced the issue. Hope to have a fix soon.

PeterJohnson, 2022-10-29:
Note this type switch behavior is no longer present in NT4.

jpokornyiii, 2022-12-01:
Updated node-ntcore to have fix, included updated version in this repo.

closed 2022-12-01 by jpokornyiii