wpilibsuite/wpilib-ws-robot-romi #114: Beep!

joshuathooyavan, 2021-12-07:
Can a beep() method of sorts be added to the OnboardIO class?

There is a buzzer onboard that I have been able to access when coding the Romi using the Arduino IDE (and no Pi), so I imagine it's possible.

jpokornyiii, 2021-12-08:
We don't currently have support for the buzzer on the Romi. One option is to add it as a custom device. We will move this to our backlog of features to support and review. Thanks.

swalberg, 2021-12-09:
Thinking out loud here...

Rather than doing things over web sockets and requiring that every piece of hardware be implemented as a custom device in the NodeJS app, is there some way of leaning on the underlying Linux OS?

As a thought, what if, when I made a `AnalogInput sensor = new AnalogInput(101)`, every time I ran `sensor.getValue()`, instead of checking the value of a pin, it read from `/tmp/analog_input_101` and returned that? Similarly, a write to a certain GPIO could write to a file or a named pipe. Then we could have a Python/shell/whatever script actually doing whatever we want on the bot.

In the OP's scenario he could use a virtual DIO pin with a simple daemon that waits for the change on the file or pipe and then makes the call to buzz (not quite clear if the buzzer is available from the Pi or if the Pi needs to talk to the AVR)