wpilibsuite/wpilib-ws-robot-romi #115: Right motor inversion doesn't work

sciencewhiz, 2021-12-21:
As reported in wpilibsuite/frc-docs#1571, the right motor isn't inverted on the Romi. Note that I haven't tested this, but looking through the firmware it looks to be true.

Here it calls the RomiMotor call to flight the right motor. https://github.com/wpilibsuite/wpilib-ws-robot-romi/blob/13dd76c887b77295c59485cd3eefc1f96ddfa27a/firmware/src/main.cpp#L331
However, the RomiMotor class isn't actually used to write the values to the motor. Instead the Servo class is used here: https://github.com/wpilibsuite/wpilib-ws-robot-romi/blob/13dd76c887b77295c59485cd3eefc1f96ddfa27a/firmware/src/main.cpp#L293

Thus the inversion would need to be implemented before the call to the Servo class in order for it to be used.

This may have been masked previously since DifferentialDrive inverted automatically, but that was removed for 2022.

zhiquanyeo, 2021-12-24:
Closing this. The servo class is only used if any of the 5 available GPIO pins are configured as PWM outputs. The motors are controlled via the RomiMotor class:

https://github.com/wpilibsuite/wpilib-ws-robot-romi/blob/13dd76c887b77295c59485cd3eefc1f96ddfa27a/firmware/src/main.cpp#L305

Thus if both motors are given the exact same value, they should spin in opposing directions

closed 2021-12-24 by zhiquanyeo

sciencewhiz, 2021-12-24:
To close the loop. The right motor is inverted on the Romi 32U4 board, so this uninverts it to act more like a typical FRC robot that is wired red to red. Thus software inversion is necessary, and the previous comment on frc-docs was wrong about not needing to invert the right motor.